FraunhoferIOSB/multisensor_calibration #18: Unable to capture a target observation

PedroS235, 2025-04-23:
First of all, congratulations on the development of this toolbox, and trying to make it easier to calibration sensors.

I have been trying to use the toolbox for calibration camera to lidar, however when capturing a target observation I keep always having the following message:
```
[ WARN]: Calibration rejected!
Mean Reprojection Error: 88.580352 px (max. threshold: 5.000000 px)
Inliers: 0 pnts (min. threshold: 10 pnts)
Removing latest observations.
```

Sometimes, I manage to get 5-6 inliers points (depending on how I place the calibration board), but not more than that. Same for the mean re-projection, there were times where I got below the 5px threshold.

I am using the same calibration board as explained in the documentation (It could be that it's now perfect to millimeter since there is human error involved). Could this be a problem?

I am running an Ouster OS0-64 (10Hz) and and RGB camera (30Hz Rolling shutter).

Below is a screenshot 

[image]

boitumeloruf, 2025-04-24:
Hi @PedroS235, thank you.

From what I see from the detections in the LiDAR data, I assume that you have also adjusted the yaml file to have a different ordering of the ArUco markers with respect to the template. Is that correct?

Have you calibrated the intrinsics of the camera beforehand and are the correct intrinsics published on the camera_info topic? 

If that is all the case, please make sure to use the same image stream from the camera for the extrinsic as you have done for the intrinsic calibration, i.e. same size.

What have you specified as image_state during launch?

PedroS235, 2025-04-24:
Dear @boitumeloruf,
> From what I see from the detections in the LiDAR data, I assume that you have also adjusted the yaml file to have a different ordering of the ArUco markers with respect to the template. Is that correct?

Yes, I have changed the markers order to be [2, 1, 4, 3].

> Have you calibrated the intrinsics of the camera beforehand and are the correct intrinsics published on the camera_info topic?

Yes, and I also tested with a different camera (also calibrated) and got the same problem.

> If that is all the case, please make sure to use the same image stream from the camera for the extrinsic as you have done for the intrinsic calibration, i.e. same size.

This is also the same size, which in my case is the 640x480.

> What have you specified as image_state during launch?

I have used the undistorted

One thing I checked is to see the number of subscribers to the camera_info topic. I noticed that it had 0. Is it only taking a single snapshot at the beginning? Or is it constantly subscribing? If it is constantly, then it could be that.

One thing I forgot to mention is that I am running this inside a docker container running jazzy distro. However, the data is coming from a humble distro, since I was not able to build it on humble. (I use the Cyclonedds for the DDS middleware)

PedroS235, 2025-04-24:
Dear @boitumeloruf,

Sorry, I meant DISTORTED. Could this come from the calibration board itself?

PedroS235, 2025-04-24:
Yes sure. Here it is. If you would like, I can also record a small dataset and share with you.

```header:
  stamp:
    sec: 1745505525
    nanosec: 322854270
  frame_id: front_left_camera
height: 480
width: 640
distortion_model: plumb_bob
d:
- -0.315894
- 0.080007
- 0.00077
- -0.000247
- 0.0
k:
- 394.55729
- 0.0
- 322.60748
- 0.0
- 394.85789
- 245.61055
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 289.77194
- 0.0
- 323.28098
- 0.0
- 0.0
- 339.87186
- 247.79107
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
```

migranram, 2025-04-25:
Hello!

There should definitively be a subscriber for the camera_info topic. Have you seen any error / warning messages in the console output of the GUI? 

Since it is inside a Docker container and communicating with another version of ROS 2, there might be a problem with the (de-)serialization of the messages. I do not think that should be the case between Humble and Jazzy, but I have had similar problems in the past. Also: Where did you check for the subscriber, within or outside the Docker container? Check in both environments.

I will take a deeper look into it.

migranram, 2025-04-28:
Hello Pedro,

I think the second problem is to blame... Compatibility between ROS2 Distros is very fragile. I think there was a major change in the ROS communication in the Iron version, if I recall correctly. 

You could try to record a rosbag in Humble, convert it to Jazzy, and test if it works that way. I have had multiple similar problems in the past.

Let me know if that works!

PedroS235, 2025-04-28:
Hi @migranram,

Yeah that is most likely the case. Anyway, thank you for your help. I will close the issue in that case.

closed 2025-04-28 by PedroS235