FraunhoferIOSB/multisensor_calibration #20: Add support for Radar sensor calibration

JSCodesTech, 2025-08-14:
To extend the applicability of this calibration framework to more real-world scenarios, it would be great to support Radar-to-sensor calibration (e.g. Radar–Camera or Radar–LiDAR).

PS: thanks for the great work on this toolkit!

migranram, 2025-08-14:
Hi!

Thanks for your issue. We do not use any radar sensor at the moment in our group, but I also think it would be great if the tool had capabilities for it!! 

Pull Requests and contribution from the open-source community are more than welcome! So feel free to send a PR with the necessary changes to enable radar- calibration. I can help you if you have any questions or problems!

PS: Depending on the radar sensor you have: Could you also try to convert its data to point clouds and use it as if it were a LiDAR? I don't know if it would work, but it would be interesting to know.

JSCodesTech, 2025-08-14:
Hi,
Thank you for your quick reply.
I am currently doing my master's degree and am working on a scientific project seminar for which I only need a camera-to-lidar calibration so far.

But for my upcoming master's thesis, I may need to add a radar sensor. So it would have been nice if it had already been included in your package. I'll see how I proceed. 

PS: interesting approach

closed 2026-02-05 by migranram